LilyGO/TTGO-LORA32 #3: Inconsistent pin mapping

fedy0, 2018-02-01:
[1] Duplicated LoRa_MOSI, No LoRa_MISO 
[2] OLED and other LoRa pin mapping are unclear due to the two new pull requests with different changes

Please update the pin out as soon as possible

kalon33, 2018-02-04:
@fedy0 I worked with the one from [Bluejedi on TTN forum](https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973/80?u=kalon33) that was OK:

[image]

Hope that it helps

fedy0, 2018-02-04:
Thanks very much for the assistance. Another question please. Does it mean that LoRa DIO1 and DIO2 pins attached to esp32 should be unused when using LoRa communication?

kalon33, 2018-02-05:
If you want to use your board for LoRaWAN communication, you need to bridge LoRa_DIO2 to GPIO 32 and LoRaDIO1 to GPIO 33 as indicated earlier in the forum thread I linked, or your program will hang after sending a packet (as I experienced), or not work. OLED-SDA and OLED-SCL are perfectly usable in I2C, and I use it successfully with a BME280 sensor.

Don't hesitate to take some time to read the forum thread there https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2 there's really plently of info about the TTGO LORA32 V2.0 board from users.

fedy0, 2018-02-08:
@kalon33 thanks

But from the link and other sources like @hallard pull request on https://github.com/matthijskooijman/arduino-lmic provides a means to either use only DIO0 or no DIO pins for LoRaWAN stack.

And in this TTGO V2, are other DIO (DIO1 and DIO2) connected to gpio12 & 11?

I ask this because it was not stated clearly in the link you shared.

robsonos, 2019-02-08:
SCL and LED_BUILTIN are wired to the same pin! That sucks.

joseluislucas, 2019-02-18:
> SCL and LED_BUILTIN are wired to the same pin! That sucks.

Why?

jinyistudio, 2019-06-28:
Does it has other DIO,ADC for iot application ?

Italque, 2019-08-04:
> If you want to use your board for LoRaWAN communication (using LMIC), you need to bridge LoRa_DIO2 to GPIO 32 and LoRaDIO1 to GPIO 33 as indicated earlier in the forum thread I linked, or your program will hang after sending a packet (as I experienced), or not work. OLED-SDA and OLED-SCL are perfectly usable in I2C, and I use it successfully with a BME280 sensor.
> 
> Don't hesitate to take some time to read the forum thread there https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2 there's really plently of info about the TTGO LORA32 V2.0 board from users.

Hi kalon33 and thanks for share this precious informations, you save my day! Could you share more informations on how to use the TTGO LORA32 V2.0  with the BME280? I tried to make it work more than once but without success. Thanks in advance!