LilyGO/TTGO-LORA32 #18: Brownout on battery --> starting bluetooth

lemmingDev, 2018-09-28:
Hi

If I start my board on battery power it keeps restarting when I call:

BLEDevice::init("ESP32");  // start bluetooth

If I start the sketch with USB cable in, it works fine
If I start it on battery power, it will bootloop due to brownout, but then if I plug the USB in, it will start working fine.

The strange thing is, if I remove the USB cable after the sketch is running fine, it will work just fine on battery power. 

This happens on TTGO LORA32 V2 - I have 2 boards and both exhibit this behaviour

On TTGO LORA V1, it works fine starting on battery and I have no issues - I have 2 boards and both are fine

Can someone please offer help?

lemmingDev, 2018-09-28:
Just checked - same issue occurs on starting wifi --> WiFi.scanNetworks();

Does having the USB cable in for a short time charge up some kind of capacitor that prevents brownout occuring?

lemmingDev, 2018-10-02:
Hardware design error?

corck, 2018-10-09:
I've the same problem as you describe, actually haven't found out whats causing it.

lemmingDev, 2018-10-21:
Doesn't look like they're willing to make a comment okn the issue...

lyusupov, 2018-10-24:
@lemmingDev : see if [this workaround](https://github.com/lyusupov/ESP32-NODEMCU-ADAPTER/issues/3) (1000uF electrolitic capacitor on 3.3V power rails) can help you.