LilyGO/TTGO-LORA32 #21: Battery Monitoring with BaT_test_ssd1306

fabioetec, 2019-01-12:
Hi,

I tried the BaT_test_ssd1306 sketch but I can't get it to work.
The value visualized on display is always 0.0000 !!!
Is pin 35 a non-exposed ADC pin where I can read voltage or should I connect something to that pin?

Thank you
Fabio

DemitasseNZ, 2019-11-21:
You need to add a voltage divider to send half of the battery voltage to the ADC on GPIO 35.
I connected a 24K resistor between GPIO 35 & ground and another 24K resistor between GPIO 35 and the battery positive terminal.
Now I get a good voltage reading.
You need two resistors of the same value and around 10K, the exact value is not so important I am just short of resistors & have two 24K metal film resistors.

Shaedo, 2023-03-05:
If this is correct, why is do they state "Add battery voltage detection Pin IO35" with the revision 5 years ago, and what's the point of the voltage dividers in the schematic? Adding a voltage divider would allow you to measure on any ADC, so what's special about GPIO 35? (I might be missing something obvious here.... ) (I appreciate this is a nero-ing the post, but it's still valid)