LilyGO/TTGO-LORA32 #23: Wich LoRa library is used?

ClimberWue, 2019-07-19:
In your code I found
#include <lora.h>
Do you have a link to the proper source or library?

morganrallen, 2019-08-19:
It can be downloaded through the Arduino Library Manager.
From the Arduino IDE...

Sketch > Include Library > Manage Libraries
Type LoRa into the search and find the one by "Sandeep Mistry"

The official repo for this library is at https://github.com/sandeepmistry/arduino-LoRa/

TrullT, 2019-09-11:
Thanks  - saved me a lot of time. It would be great if this Library metadata was recorded in a comment thusly:
#include <LoRa.h> // https://github.com/sandeepmistry/arduino-LoRa/ or Arduino Library Manager

morganrallen, 2019-09-11:
I agree and hope to work with LilyGO in the future to make these sort of things more clear.

jbrepogmailcom, 2020-01-28:
Hello, what library do you recommend for LoRaWAN? Will the ibm-lmic library work without pin redefinition?