LilyGO/TTGO-LORA32 #28: TTGO LoRa v1.0 high deep sleep current consumption

mbobinger, 2020-08-28:
Dear TTGO-Team,

I have purchased the v1.0 TTGO ESP32 & SX1276 version here:
https://www.banggood.com/2Pcs-LILYGO-TTGO-ESP32-SX1276-LoRa-868MHz-bluetooth-WI-FI-Lora-Internet-Antenna-Development-Board-p-1295045.html?cur_warehouse=CN&rmmds=search

The deep sleep current consumption is 12mA when powered over battery and 14mA when powering the 3.3V rail with a benchtop supply.

I'm quite sure I've done nothing wrong.

Can you recommend how to lower the power consumption? Are there new boards with better efficiency?
Best and thank you!

cuplesoft, 2020-09-28:
Same problem here..
I bought 2 boards.
Current consumption in deep sleep mode is 11-12mA.

karlvoigt, 2024-12-09:
Did you ever manage to figure out the issue or find a lower power alternative. As I am stuck on a project where I need very low power consumption in sleep mode and this high consumption is killing me.

martin3000, 2024-12-09:
I bought a TinyPico board and the power comsumption was still too high. So I gave up my project. ESP32 ist not for battery.

MiguelAngelLV, 2025-04-14:
> I bought a TinyPico board and the power comsumption was still too high. So I gave up my project. ESP32 ist not for battery.

The problem is not the ESP32, it is the board. I have some ESP32 boards (as firebettlee) with 12uA power consumption in deep sleep