LilyGO/TTGO-LORA32 #29: LoRa.onReceive is not working

Nargott, 2020-09-14:
Hi. When I'm used _LoRa.receive()_ in setup and _LoRa.parsePacket()_ in the loop section, packets come as expected. But I want to use esp_light_sleep_start to save power. But I didn't receive any callbacks even without power savings. Of course, LoRa configured as _LoRa.setPins(SS, RST, DI0)_, where _#define DI0 26_

Is anyone has a positive experience using _LoRa.onReceive_?

beaglehardware, 2021-01-10:
I had the same problem, but finally figured out it was not a 3 band board as advertised. My board runs only on 433Mhz.

roseeng, 2022-08-12:
I don't see how 433MHz vs 868MHz could affect whether or not the callbacks work. I have the same issue as @Nargott, receive works in the loo(), but not as a callback.

If I explicitly configure the DI0 pin as input using ```pinMode(DI0, INPUT)```, I get exactly one callback, but the the board craches an restarts. I tried this since I have seen other people mentioning it as a possible solution.