LilyGO/TTGO-LORA32 #32: LoRa32 TTGO T3 v1.6.1 Charging?

WaGi-Coding, 2022-08-11:
So I wonder if i have to have the battery ON/OFF switch set to on or off in order to charge the battery while powered via USB. I am not even sure if it's capable of charging a 18650 plugged into the battery socket.

Can somebody explain? I currently turn the switch to OFF when i power it via USB, but i am not sure if it's charging in that state.

Also, unfortunately, it's always running when connected via USB, but thats just a little issue i've with it. Wasn't this about to be changed with 1.6.1, at least according to an issue about the 1.6, it was said that this behaviour will change with the 1.6.1, but seems it did not.

Tyler-2, 2022-12-28:
I am not particularly good at any of this, but my understanding is that the on/off switch simply disconnects the battery. When it's off, it will still run from the USB port. When it's on, it will run from the battery or USB port while also charging the battery.

AdamM68, 2025-03-09:
it is high time to answer these questions .  Is there somewhere any proper explanation how charging and on/off switch are working ?

Tyler-2, 2025-03-09:
I stand by my previous post.