LilyGO/TTGO-LORA32 #37: Inconsistent pinout TTGO LoRa T3-S3

BadCactus634, 2024-04-22:
Hi, I'm trying to attach a BME280 sensor to my Lilygo TTGO LoRa T3-S3 that I just bought, but the pinout on the site is not correct.
[image]
As you can see, the pins of the I2C bus (SDA, SCL) are documented as pin 18 and 17, but these pins doesn't exist on the pinout....
where should I connect my I2C sensor?

Grey-Lancaster, 2024-05-06:
17 and 18 are used for the oled.  You can use other pins
https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/io_mux.html#:~:text=Now%2C%20for%20the%20ESP32%2C%20the,need%20to%20change%20the%20pins.

ryziopl, 2025-07-04:
So to connect AHT20 Temp sensor with SDA and SCL pins witch i need to use becouse none of them work for me with Mesthastic.

Grey-Lancaster, 2025-07-04:
try 45 and 46