LilyGO/TTGO-LORA32 #38: How to use the 2 SPO buses (LoRa & SD Card) ?

F5OPV, 2024-06-03:
Hello everybody,
Is there any way to get SPI LORA bus and SPI SDCARD bus working simultaneously with 2 SPI instances?

RPdenBoer, 2024-06-06:
This seems to work for me. Using standard SD library and https://github.com/sandeepmistry/arduino-LoRa.git on LoRa32 V2.1_1.6 devboard.
SD and LoRa libraries have different ways of passing in the SPI bus you want to use; see comments. 

```
#include <Arduino.h>

#include <SPI.h>
#include <LoRa.h>
#include <SD.h>

// Define pins for SD card 
#define SD_SCK 14
#define SD_MISO 2
#define SD_MOSI 15
#define SD_CS 13

// Define pins for LoRa 
#define LORA_SCK 5
#define LORA_MISO 19
#define LORA_MOSI 27
#define LORA_CS 18
#define LORA_RST 23
#define LORA_IRQ 26

// Define SPI buses (can be swapped)
SPIClass spiSD(VSPI);
SPIClass spiLoRa(HSPI);

// Init flags
bool sdInitialized = false;
bool loraInitialized = false;

void setup() {
    Serial.begin(115200);

    // Initialize SD card SPI 
    spiSD.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

    // Pass in SD-SPI-bus object here as second argument
    if (!SD.begin(SD_CS, spiSD)) {
      Serial.println("SD Card Mount Failed");
    } else {
      Serial.println("SD Card initialized");
      sdInitialized = true;
    }

    //Initialize LoRa SPI 
    spiLoRa.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_CS);
    LoRa.setPins(LORA_CS, LORA_RST, LORA_IRQ); // Set LoRa pins

    // Pass in LoRa-SPI-bus object here as single argument
    LoRa.setSPI(spiLoRa);

    if (!LoRa.begin(915E6)) { // Use your frequency
      Serial.println("Starting LoRa failed!");
    } else {
      Serial.println("LoRa initialized");
      loraInitialized = true;
    }
}

void loop() {
  // Example: Write data to SD card
  if (sdInitialized) {
    File file = SD.open("/test.txt", FILE_WRITE);
    if (file) {
      file.println("Hello from SD card");
      file.close();
      Serial.println("Data written to SD card");
    } else {
      Serial.println("Failed to open file for writing");
    }
  }

  // Example: Send data via LoRa
  if (loraInitialized) {
    LoRa.beginPacket();
    LoRa.print("Hello from LoRa");
    LoRa.endPacket();
    Serial.println("Data sent via LoRa");
  }

  delay(2000);
}
```

F5OPV, 2024-06-06:
Thanks a lot for your answer and code
I will try it ASAP.
Best regards
F5OPV

F5OPV, 2024-06-07:
Hi,

I tested your code this night but it seems there's an issue.
Only the first record in the txt file is writen on sd and following ones not .

Any idea abt the reason of this issue ?

F5OPV

RPdenBoer, 2024-06-08:
That is expected behaviour for the code. Please see the SD library documentation to understand; discussion regarding it doesn't really belong here.

To get you started though, you could open the file in setup and never close it - use SD.flush() in your loop instead. You could alternatively/also substitute FILE_READ with FILE_APPEND in SD.open()

F5OPV, 2024-06-08:
Thanks for answer and your precious code.
Solved the issue by replacing the FILE_WRITE by FILE_APPEND.

And now your code works like a charm !
Thanks agn for help.

73

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

closed 2024-06-09 by F5OPV